Caribou3d/CaribouMK3s-Firmware #16: Extruder steps

TigerClaw33, 2019-10-04:
Hi, 

I don't know if you tested it. I am using the Mod with Mosquito Hotend but without the Bondtech Extruder. The steps in the firmware have the parameter 280.00. For my configuration (maybve because the capricorn tube from the Mosquito) I have to use exactly 301.90 as parameter.

3d-gussner, 2019-10-04:
Hi @TigerClaw33 
Thanks for the issue report.
The E-Steps in the firmware are an average value. We just cannot test every possible mod is available, so I have to say we didn't test your specific configuration.

If somebody wants to dial in a printer correctly/more precise, we advice everyone to calibrate their machine to their specific hardware.
In your case you have a mod without the Bondtech AND without the E3Dv6 hotend, so it always makes sense to calibrate your extruder yourself. Wel done BTW.

How many times did you run the calibration procedure?
My experience is to do it at least 3 time better 5 or more and get an average value of the tests, as it is quite hard to manually measure precisely with calipers and your eye.

TigerClaw33, 2019-10-04:
Hi, 

I run it 5 times. At the beginning everything was ok. I printed PEEK and after that I noticed that I have an underextrusion. 

So I recalibrated it. I checked it also with other nozzles. Now it seems to be working. O think I have some peek material inside the heartbreak ... Or maybe I had other values working before because the hotend was new ... Who knows ;)

3d-gussner, 2019-10-04:
@TigerClaw33 
Wow PEEK ... glad to hear that you got it right, pity you had to make this experience with this very non-in-expensive material.
Would be interesting if it is a PEEK thing only or this new value suites also other materials as well. Then we could point it down to your specific combo of hardware. If it is not then it is something PEEK specific (filament diameter), and then it would make more sense to have an E-Step value that works for most filaments and use flow/extrusion multiplier for PEEK.

TigerClaw33, 2019-10-04:
I calibrated it with PLA filament bit after printing PEEK. Therefore I tested different nozzles. Maybe the value is because I am using a Capricorn tube and is has a lower diameter than standard PTFE tubes.

3d-gussner, 2019-10-06:
Is it okay if i close that issue? Or can you do it?

closed 2019-10-06 by TigerClaw33