Caribou3d/CaribouMK3s-Firmware #17: Error ArduinoIDE

TigerClaw33, 2019-10-09:
Hi, 
if I compile your firmware I get this error:
```
Archiving built core (caching) in: C:\Users\admin\AppData\Local\Temp\arduino_cache_816924\core\core_rambo_avr_rambo_0c812875ac70eb4a9b385d8fb077f54c.a
wiring.c.o (symbol from plugin): In function `__vector_23':

(.text+0x0): multiple definition of `__vector_23'

sketch\heatbed_pwm.cpp.o (symbol from plugin):(.text+0x0): first defined here

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

collect2.exe: error: ld returned 1 exit status

exit status 1
Fehler beim Kompilieren für das Board RAMBo.
```

If i compile the one from prusa I get this error:
```
sketch\Marlin_main.cpp: In function 'T gcode_M600_filament_change_z_shift()':

Marlin_main.cpp:2991: error: static assertion failed: Z-range too high, change the T type from uint8_t to uint16_t

  static_assert(Z_MAX_POS < (255 - FILAMENTCHANGE_ZADD), "Z-range too high, change the T type from uint8_t to uint16_t");

  ^

exit status 1
static assertion failed: Z-range too high, change the T type from uint8_t to uint16_t
```

Do you have any idea?

3d-gussner, 2019-10-10:
Hi @TigerClaw33 ,

did you follow the [instructions](https://github.com/Zaribo/Zaribo-Firmware#windows) ? Especially to change the `LANG_MODE` [see 2.c.](https://github.com/Zaribo/Zaribo-Firmware#2-source-code-compilation)
To compile Zaribo/Prusa firmware please use [Arduino IDE 1.8.5](https://www.arduino.cc/en/Main/OldSoftwareReleases#previous) and DO NOT EVER update the "AVR Boards" in the Board Manager.
Even better is to use Arduino IDE 1.8.5 in [portable mode](https://www.arduino.cc/en/Guide/PortableIDE) as default Arduino IDE installation uses shared folders so other/newer data could cause issues.

I wrote the `PF-build.sh` script because many had issues to have a known stable build environment.
Just follow these instructions [Linux Subsystem for Windows](https://github.com/Zaribo/Zaribo-Firmware#using-linux-subsystem-under-windows-10-64-bit) or [Git-bash](https://github.com/Zaribo/Zaribo-Firmware#using-git-bash-under-windows-10-64-bit)

The Zaribo firmware has some additional fixes and code to ensure that it works on Zaribo/Bear and modded Prusa printers:
- Bugfix for motor orientation
- Unlimited height ... That's why you get the `static assertion failed: Z-range too high, change the T type from uint8_t to uint16_t`
- Extruder dimension based on Prusa Extruder body R3 and NOT Prusa R4 or R5
  - The Prusa R4/R5 extruder body is bit shorter and Zaribo as well Bear decided to stay at R3 revison dimension.
- Dutch language support
- Slice High Temperature Thermistor support

3d-gussner, 2019-10-11:
@TigerClaw33 How is it going? Did you get any closer to compile your own firmware?

3d-gussner, 2019-10-14:
@TigerClaw33 Any update?

3d-gussner, 2019-10-16:
As there is no update I close the issue.

closed 2019-10-16 by 3d-gussner

Demolux, 2020-02-22:
Hi!
Can you explain how you achieved unlimited height? I want to mod the Prusa FW to accept a height of 500. It would be awesome you can tell me how to change uint8_t to uint16_t.