Caribou3d/CaribouMK3s-Firmware #18: git

lijinlin19970813, 2019-10-17:
I used git to follow your steps, but I got an error:PF-build.sh: line 449: /dev/fd/62: No such file or directory，And it is empty in the Prusa-Firmware-build folder，why?

3d-gussner, 2019-10-17:
@lijinlin19970813 I saw that few days ago during an one-on-one session. Have to try to figure out what happend since I wrote the instructions. It can be that git-bash newer version cause the issue, can't tell yet.
Temporary solution is to use Linux Subsystem for Windows with Ubuntu.

lijinlin19970813, 2019-10-17:
@3d-gussner You can use linux to run build.sh to generate hex files? Is it in accordance with the steps in the README?

3d-gussner, 2019-10-17:
@lijinlin19970813 Yes and no, depending on your Linux. Which Linux distribution do you use? I have tested it with Ubuntu. If there is the need I may consider to get it running on other distributions as well.

`PF-build.sh` is based on `build.sh` and both run on Linux, the `PF-build.sh` is written by me and others to make it hopefully bit easier to compile the firmware.

`Git-bash`  is a Linux for Windows see http://www.mingw.org/

I have mingw64 in version: 
```
$ git --version
git version 2.22.0.windows.1
```

Which did you install?
Can you post the results of `$ ls /dev/`

3d-gussner, 2019-10-23:
@lijinlin19970813 I will close this issue due to no update/answer. If you still have a question please feel free to reopen that issue.

closed 2019-10-23 by 3d-gussner