Caribou3d/CaribouMK3s-Firmware #26: 3.9.0 BETA Support

dennisbappert, 2020-02-03:
Prusa recently released 3.9.0 BETA. It would be great to have releases available for the BETA version to get the benefits of LA1.5.

Best, Dennis

3d-gussner, 2020-02-04:
@dennisbappert Hi Denis thanks for opening this issue. As it is a Beta where things can go wrong we decided not to make it public and send few people the Beta Firmware. I think that FW 3.9.0-RC1 will come soon and we will publish this as soon as possible.

dennisbappert, 2020-05-16:
Still no updates?

3d-gussner, 2020-05-16:
Sorry, we forgot to update this issue. There is a new branch for FW3.9.0, which been tested by few Zaribo Facebook members.

3d-gussner, 2020-05-16:
@dennisbappert Please checkout https://github.com/Zaribo/Zaribo-Firmware/tree/Zaribo-MK3-FW3.9.0
Also hitting the github watch could help you to see if something new has been posted.

dennisbappert, 2020-05-17:
Thank you very much, are there precompiled binaries available for 3.9.0, too?

3d-gussner, 2020-05-17:
The final will be there very soon and we will provide these. Please stay tuned.

3d-gussner, 2020-05-21:
@dennisbappert final is there. I will close this issue.

closed 2020-05-21 by 3d-gussner