Caribou3d/CaribouMK3s-Firmware #27: XYZ- calibration fails with FW390-Zaribo-MK25S-220-Build4153-DE

brinkeb, 2020-06-02:
Someone had a problem with "FW390-Zaribo-MK25S-220-Build4153-DE". XYZ- calibration stopped at calibration point "1". Prusa firmware works, only changed Zmax to 220.

3d-gussner, 2020-06-03:
Please collect as much information as possible.
Also please fill in the form and don't overwritte it. 
In the future we will close these issues where the form information is missing.

3d-gussner, 2020-06-03:
@brinkeb Can you ask the person with the issue if she/he would like to help with the investigation here?
I would like to try few things:
- Home xyz with `G28W`
- Move z axis be knob to see when exactly his printer hits the Z-top holders

3d-gussner, 2020-06-03:
There might be an issue with
https://github.com/prusa3d/Prusa-Firmware/compare/MK3...Zaribo:Zaribo-MK3#diff-df5e16bf433c533f96a37bdb3456d6c9R2191-R2205
and/or
https://github.com/prusa3d/Prusa-Firmware/compare/MK3...Zaribo:Zaribo-MK3#diff-df5e16bf433c533f96a37bdb3456d6c9R3076
and/or
https://github.com/prusa3d/Prusa-Firmware/compare/MK3...Zaribo:Zaribo-MK3#diff-68a5b34f911d0dfba612ae98d305facaR3520-R3534

brinkeb, 2020-06-03:
> @brinkeb Can you ask the person with the issue if she/he would like to help with the investigation here?
> I would like to try few things:
> 
> * Home xyz with `G28W`
> * Move z axis be knob to see when exactly his printer hits the Z-top holders

I´ve asked him. Waiting for an answer now.

3d-gussner, 2020-06-08:
@brinkeb Still no feedback from user? If not I will close this issue and next FW she/he will have to report back.

brinkeb, 2020-06-08:
Finally got feedback from him. Seems, that his Zmax is 225 and he needs custom firmware, actually. So, I don´t think that this a bug in FW3.9.0. But there is still no explanation why Zaribo- firmware 3.9.0 with Zmax = 220 fails calbration while Prusa- firmware (set to Zmax= 220) passes calibration.

closed 2021-07-12 by wschadow