Caribou3d/CaribouMK3s-Firmware #30: [BUG]<MK3s-220Upgrade-Extrude-Fehler-über-usb>

spacy2878, 2020-07-20:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3s-Zaribo-220-Upgrade
**Printer firmware version**-  FW390-Zaribo-MK3S-BE-220-Build4153

**Describe the bug**
  Wenn man in Pronterface auf "Extrude" geht sollte die eingestellte menge extrudiert werden. das wird auch getan.. aber nur 1mal... wir mal erneut extrudieren muss man zuerst "Reverse" drücken.. dann geht wieder 1mal Extrudieren.... dieser Fehler ist da in der Zaribo FW390 genauso wie in der Prusa3,9,0. in der vorrigen FW 381 funktioniert alles einwandfrei.

3d-gussner, 2020-07-20:
@spacy2878 Kannst Du bitte die Terminalausgabe posten?

Versuche auch bitte etwas kleines über pronterface zu drucken um zu sehen ob es ein allgemeines Problem ist mit USB commands.

Versuche auch mal den Durcker vorzuheizen und manuel Gcode auszuführen:

```
M83 //relative mode
G1 F50 //feedrate 50mm/min
G1 E50 //extrude 50mm
G1 E50 //extrude 50mm

```

spacy2878, 2020-07-20:
habs grad ausprobiert... bei G1 E50 ist es genauso.. 1mal gehts. ein 2.mal erst nach Reverse... bei der letzten FW funktionierts einwandfrei.. drucken über Pronterface funktioniert dagegen bei beiden FW einwandfrei.... das ist sowas von merkwürdig....

 

 
 

Gesendet: Montag, 20. Juli 2020 um 17:44 Uhr

spacy2878, 2020-07-20:
hab noch bisschen probiert... also m83 g1 f50 etc... funktioniert... hab nochmal mit dem extrude button probiert... wenn ich M83 eingebe und dann Extrude drücke geht es... auch mehrmals hintereinander.. wenn ich dann resette und extrude allein drücke... kommt der Fehler... aber Extrude sollte ja ohne m83 gehen...

 

 

Gesendet: Montag, 20. Juli 2020 um 17:44 Uhr

3d-gussner, 2020-07-21:
Ich habe es mal selbst probiert und bekomme etwas andere ergebnisse:
- In Pronterface kann ich das erst mal 10mm extruden.
- beim zweiten Versuch jedoch passiert überhaupt nix, also KEIN unload oder drehen vom extruder.

Mit Octoprint geht es mit beiden FW versionen da Octoprint
```
M83 //set extruder to relative mode
G1 E10 F300 //extrude 10mm with 300mm/min
M82 //set extruder to absolute mode
```
ausführt.

Habe das mal bei Prusa gepostet und warte noch ne Antwort ab.
Mit FW3.8.1 -> FW3.9.0 wurden paar Verbesserungen hinzugefügt:
- Auch wurde ein Teil "zurück" genommen um mit FW3.8.1 übereinzustimmen.

spacy2878, 2020-07-21:
In Pronterface kann ich das erst mal 10mm extruden.
    beim zweiten Versuch jedoch passiert überhaupt nix, also KEIN unload oder drehen vom extruder.
 das ist bei mir genauso.,,,, erst wenn ich selber REVERSE anklicke funktioniert ein erneutes Extruden

3d-gussner, 2020-07-21:
Ich habe es so verstanden, dass wenn ein zweites mal extrude gewählt wird der ein unload macht.

Dann macht es sinn. Der Drucker ist mit FW3.8.1 in relative mode `M83` und in FW3.9.0 in absolute mode `M82` nach dem einschalten.

Schauen wo der "revert", der das Problem eigentlich hätte lösen sollen, evtl. miss gegangen ist.

spacy2878, 2020-07-21:
das würde auch erklären warum bei fw 3.9.0 es bei pronterface mit dem extrude Button funktioniert nachdem ich M83 eingebe....

spacy2878, 2020-08-05:
keiner ne idee was da defekt ist an der FW?!?!?!?!#

closed 2021-07-12 by wschadow

spacy2878, 2021-07-12:
Top wenn fehler nie behoben..nur ignoriert und einfach gelöscht wird... wie letztes update. Wo die deutsche version nicht ging....es hies nur...wie kpmmern uns drum.... seither totgeschwiegen.... .an erfährt auch nur per zufall das die palette, für sie man massig kohle vorstrecken muss...um weitere monate verschoben wird.--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

   
  Closed #30.

wschadow, 2021-07-12:
Die Fehlermeldung bezieht sich auf 3.9.0. Wir haben heute 3.10.0 veröffentlicht. Dort funktioniert auch die Sprachumstellung. Mosaic hat Corona bedingt die Auslieferung um 2 Monate verschoben. Darauf haben wir keine Einfluss.