Caribou3d/CaribouMK3s-Firmware #35: [BUG] XYZ- calibration fails with FW3.10.0 MK3S 220

Leuprochon, 2021-05-10:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]

MK3S 220

**Printer firmware version**-  [e.g. 3.8.1, 3.8.1-RC1, ...]

FW3.10.0

**Describe the bug**

When I start the XYZ calibration the pinda doesn't find the bed point. It's impossible to calibrate my printer with XYZ calibration menu. So I reinstal the older firmware (3.9.3) with wich I don't had this issue.

3d-gussner, 2021-05-10:
@Leuprochon Which MK3S 220 do you have?
8mm rods
Bondtech

wschadow, 2021-07-12:
It's a bug in the initial release of 3.10.0. Please try the new release.

closed 2021-07-12 by wschadow