Caribou3d/CaribouMK3s-Firmware #38: [BUG]<Enter comprehensive title>

iovsiann, 2022-05-05:
**Printer type** - Caribou 420 (upgrade from Prusa MK3S), 
**Printer firmware version** - 3.11.0-5786, /00_Latest Version 3.11.0/Caribou420/MK3S/Caribou/FW3110-Caribou420-MK3S-Build5786-EN.hex
**MMU upgrade** - no MMU
**SD card or USB/Octoprint** - SD card

**Describe the bug**
Y axis is inverted. Prints come out mirrored.
When a print finishes and it's time to remove the finished print, the hot bed slides away from the user - not towards the user.
However, the initial setup, self-test and calibration (including XYZ) all pass fine.

**To Reproduce**
1) Upload firmware file /00_Latest Version 3.11.0/Caribou420/MK3S/Caribou/FW3110-Caribou420-MK3S-Build5786-EN.hex using Prusa Slicer firmware upload.
2) Power up the printer and follow LCD prompts to complete self-test and calibration
3) Print something

**Expected behavior**
Y-axis should operate normally
  
**G-code**
The G-code prints number "6", but the print comes out mirrored.
[Mirror test 6_0.2mm_PETG_MK3S_12m.zip](https://github.com/Caribou3d/CaribouMK3s-Firmware/files/8634576/Mirror.test.6_0.2mm_PETG_MK3S_12m.zip)

**Video**
[image]

https://user-images.githubusercontent.com/9521508/167009651-40dcb20e-bc9c-4c67-a1ea-c9097f605a8a.mp4

wschadow, 2022-06-17:
What was the behavior before? Is this a new build? Have you checked the y-belt tensioner being in the right position?

iovsiann, 2022-06-27:
The printer was built in January 2021, so no - this is not a new build.

The behavior was correct before loading 3.11.0-5786 firmware; however, the previous firmware (that worked correctly) was Prusa MK3S hand-modified to support 420mm (the printer was back when Caribou 420mm firmware binaries were not yet available, so I had to compile firmware myself).

The y-belt tensioner seems fine. It worked correctly before loading the latest firmware.

closed 2022-07-04 by wschadow