Caribou3d/CaribouMK3s-Firmware #39: Text line on LCD moves across the screen at warp speed.

Bladevane, 2023-02-25:
**Printer type Caribou MK3S 320
**Printer firmware version 3.12.1

**MMU upgrade** - [e.g. MMU2S, MMU2, MMU1]
**MMU upgrade firmware version** - [e.g. 1.0.6, 1.0.6-RC2, ...]

**SD card 
  
**Describe the bug**
  After upgrading to 3.12.1 and installing the printed Mirrored Y Belt Holder the calibration of the nozzle temperature went well. When printing the text line at the bottom of the LCD flies by at warp speed during initial bed level checking and during the printing process.
  
**To Reproduce**
  Please describe steps to reproduce the behavior.
  
**Expected behavior**
  A clear and concise description of what you expected to happen.
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.

wschadow, 2023-03-02:
Thanks for the report. It's fixed in 3.12.2.

closed 2023-03-02 by wschadow

Bladevane, 2023-03-03:
Great, thanks. Speedy response, as ever.

 

Regards,

 

Glyn.